Xilinx/embeddedsw #325: AXI-Stream Fifo C-driver missing from generated platform in Vitis Unified

digitalnumeric, 2024-12-02:
We have tried using AXI-Stream FIFO but it is not generating the C-driver code (i.e. xllfifo etc.) in the platform. 
We have tried it with both Vitis Unified 2023.2 and 2024.2 and the problem exists in both.

Here are several forum posts raising the issue:
https://adaptivesupport.amd.com/s/question/0D52E00006hpX9RSAU/help-fatal-error-xllfifoh-no-such-file-or-directory?language=en_US
https://adaptivesupport.amd.com/s/question/0D54U00008Ri24SSAR/axistream-fifo-driver-not-recognized-vitis-unified-ide-20232?language=en_US
https://adaptivesupport.amd.com/s/question/0D54U00008sLBuMSAW/how-to-use-the-axi-stream-fifo-drivers-xllfifoh-xllfifogc-etc-in-the-project?language=en_US

anirudha1977, 2024-12-03:
Hi.. Thanks for reporting this.
We have got this issue reported from internal sources and already have sent a fix for internal builds..
We will think if we need to create an AR for this ..
Sorry for the trouble..

digitalnumeric, 2024-12-03:
Sounds good. Thank you.

digitalnumeric, 2025-02-19:
This forum post has the workaround: 

[Forum Post with Solution](https://adaptivesupport.amd.com/s/question/0D54U00008sLBuMSAW/how-to-use-the-axi-stream-fifo-drivers-xllfifoh-xllfifogc-etc-in-the-project?language=en_US)