Xilinx/embeddedsw #326: Vitis 2024.2 is missing CMake definition for __FILENAME__ used in xil_assert.h in some cases

digitalnumeric, 2024-12-19:
Using Vitis 2024.2, when performing a build we noticed many-many warnings about \_\_FILENAME\_\_ not being defined.

Looking at xil_assert.h in github here:
https://github.com/Xilinx/embeddedsw/blob/master/lib/bsp/standalone/src/common/xil_assert.h

It appears that the change was made expecting \_\_FILENAME\_\_ to be defined by CMake but based on all the build warnings, it seems like that is not happening in all cases.

Please see the forum post below that includes logs showing a sampling of the warnings that also shows it is being defined sometimes via -D__FILENAME__. 

https://adaptivesupport.amd.com/s/question/0D54U00008uNQCRSA4/missing-cmake-definition-for-filename-used-in-xilasserth-in-some-cases

anirudha1977, 2024-12-19:
Thanks for reporting this.
We got to know about this issue from a different source few days back and have prepared a fix which is being tested internally. The fix should be available in 2025.1.
If you need to the fix right away, we can send it once it is internally validated.

kedareswararao, 2024-12-19:
Hi @digitalnumeric ,

Please follow the below steps and let us know the test result

1. git clone https://github.com/Xilinx/embeddedsw.git
2. cd embeddedsw
3. git checkout -b xlnx_rel_v2024.2 remotes/origin/xlnx_rel_v2024.2
4. git am 0001-esw-Fix-issues-with-the-__FILENAME__-macro.patch

Use the embeddedsw folder as a local repo in the Vitis.
[0001-esw-Fix-issues-with-the-__FILENAME__-macro.patch](https://github.com/user-attachments/files/18200596/0001-esw-Fix-issues-with-the-__FILENAME__-macro.patch)

digitalnumeric, 2024-12-20:
I used your steps it and appears to have fixed the \_\_FILENAME\_\_ issue we were seeing.
I used the attached Python script to perform the build.
Filename: vitis_create_full_workspace_2024.txt which needs to be renamed to vitis_create_full_workspace_2024.py to run it.
I attached the console output from running the script. 
Filename: 2024-12-20_Vitis_2024_2_Hang_after__FILENAME__patch_applied.txt
 
Now that the \_\_FILENAME\_\_ issue is fixed there are still two issues remaining keeping mefrom being able to adopt Vitis 2024.2:
 
Vitis 2024.2: CMake Policy CMP0010 is not set: Bad variable reference syntax is an error.
https://github.com/Xilinx/embeddedsw/issues/327

Vitis 2024.2: Fails to complete build, hangs
https://github.com/Xilinx/embeddedsw/issues/328

[vitis_create_full_workspace_2024.txt](https://github.com/user-attachments/files/18215100/vitis_create_full_workspace_2024.txt)
[2024-12-20_Vitis_2024_2_Hang_after__FILENAME__patch_applied.txt](https://github.com/user-attachments/files/18215101/2024-12-20_Vitis_2024_2_Hang_after__FILENAME__patch_applied.txt)

closed 2024-12-20 by digitalnumeric