Xilinx/embeddedsw #333: HWICAP global arrays have wrong size.

smartinou, 2025-03-11:
In xhwicap.c:

- l.108 series_ultra_idcodes[] is declared/defined with a size of NUM_ULTRA_SERIES_IDCODES defined to 13, while the array has a size of 14 elements.
- l.114 series_ultra_plus_idcodes[] is declared/defined with a size of NUM_ULTRA_PLUS_SERIES_IDCODES defined to 12, while the array has a size of 30 elements.

This was flagged by compiler while attempting to compile with mb-g++ (os compiler option set to mb-g++ instead of mb-gcc). It is also trivially verifiable by mere code inspection.

This setting also flags many more errors in other drivers (currently not possible to compile a typical standalone bsp with mb-g++).

Coding-xinxiangwanji2017, 2025-03-11:
Hi，您好：    我收到了您邮件，等我看完邮件以后，第一时间给您回复。

NavaKishoreManne, 2025-03-12:
Hi @smartinou,

We have already resolved this issue in our internal repository, and the fix will be included in the GitHub repository as part of the 2025.1 release branch soon. Thank you for your patience.