Xilinx/embeddedsw #336: HWICAP id code arrays missing entries.

smartinou, 2025-03-19:
in file hwicap.c, id code table series_ultra_plus_idcodes is missing some entries, as per UG570 (v1.20.1) March 4, 2025
UltraScale Architecture Configuration document, table Table 5: JTAG and IDCODE for UltraScale Architecture-based FPGAs on p.20.

For instance,  AU15P (X4ac2093) is missing.

I'm counting 29 Ultrascale+ devices (5 artix, 7 kintex, 17 virtex). Yet the array size is 28 (with wrong array size specified as per https://github.com/Xilinx/embeddedsw/issues/333) , so the AU15P looks to be missing from the array.

Sadly, this is the one part we are using...

Layout like this could help quickly identifying supported devices rather than cross-referring to the ug:

`static u32 series_ultra_plus_idcodes[] = {
    0x4A63093, // KU3P
...
};
`

Botnic, 2025-03-26:
I hope they get it fixed for the next version (so they promised)

https://adaptivesupport.amd.com/s/question/0D54U00008oA9dgSAC/bugreport-arraylength-of-idcodes-in-xhwicap-driver-wrong

sivadur, 2025-03-29:
@smartinou thanks for reporting, we will check and fix in our next release.