Xilinx/embeddedsw #339: Infinite loop based on hardware behavior

jwolfm98, 2025-04-09:
https://github.com/Xilinx/embeddedsw/blob/0f463342b5c42ff17690d396fb74381fb13168ab/ThirdParty/sw_services/lwip220/src/lwip-2.2.0/contrib/ports/xilinx/netif/xaxiemacif_physpeed.c#L234

digitalnumeric, 2025-04-09:
Not sure if this will help as we are still stuck with Vitis Unified 2023.2 and lwIP 213 and are using a KR260 board bare-metal/FreeRTOS.

We found that auto negotiation always failed unless we connected all the Ethernet connections to a switch.  Then it would succeed.  We ended up changing to a fixed 1000 speed rather than rely on auto negotiation.  Later, we found out that the KR260 board has a shared MDIO line which is what was causing the issues.  The PetaLinux driver knows how to deal with this, but lwIP does not.

venodela, 2025-04-11:
We added support for the common MDIO handling in lwip for kr260 in 2025.1 release.