Xilinx/embeddedsw #350: lwip does not function when link speed is 100 Mbps

mike919192, 2025-04-28:
There seems to be a problem with the new SDT ethernet drivers when the negotiated link speed is 100 Mbps.  I first ran into the problem with the FreeRTOS TCP Zynq port, but I reproduced the problem using the baremetal lwip [echo example](https://github.com/Xilinx/embeddedsw/tree/master/lib/sw_apps/lwip_echo_server).

From what I can tell there is a difference in behavior in the function [SetUpSLCRDivisors](https://github.com/Xilinx/embeddedsw/blob/master/ThirdParty/sw_services/lwip220/src/lwip-2.2.0/contrib/ports/xilinx/netif/xemacpsif_physpeed.c#L965).  In the old drivers this function wrote the correct divider values to the `SLCR_GEM0_CLK_CTRL_ADDR` register.  With the new drivers this function does nothing, the entire function body is behind the `#ifndef SDT`.  

I think the intention is that it is not necessary to write to this register with the SDT drivers.  And that does work with 1Gbps link speed.  But with the SDT drivers, 100Mbps link speed does not work without writing to the `SLCR_GEM0_CLK_CTRL_ADDR` register.

Here is a bit long discussion we had over on the FreeRTOS [forums](https://forums.freertos.org/t/freertos-tcp-in-amd-xilinx-vitis-2024/22809).  For now I am assuming that to workaround the problem that it is still necessary to write to the `SLCR_GEM0_CLK_CTRL_ADDR` register with user specified values in order to use 100Mbps link speed.

yobniares, 2026-09-25:
I have exactly the same issue, on standalone Zynq 7020 (emacsps, RTL8211F phy) with the new vitis 2025.2. Before that i used the 2020 version without the SDT and speed settings worked just fine. Going to find a hotfix

HariniKatakamX, 2026-09-28:
Hi, the clock register update is still required in SDT flow. However, it should be handled manually by the user based on their design. For Zynq US+ MPSoC, we have baremetal clock framework support and this takes care of the SLCR register write automatically based on the negotiated speed. But this framework support is not available for all devices and may be extended in the future. This is documented under known issues here:
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842366/Standalone+LWIP+library#Known-issues/Limitations
https://xilinx-wiki.atlassian.net/wiki/spaces/A/pages/18842012/Standalone+Ethernet+Driver
If you are using Zynq7020, the fix would be to use the SLCR register offsets provided here (https://github.com/Xilinx/embeddedsw/blob/master/ThirdParty/sw_services/lwip220/src/lwip-2.2.0/contrib/ports/xilinx/include/netif/xemacpsif.h) and re-use the SetupSLCR function with the divisors for your own design. XPAR_xx definitions are not exported in SDT flow either.